Nitrokey/nitrokey-pro-firmware #9: Unlocking password safe fails when compiling with recent gcc versions

MaVo159, 2016-07-30:
Sending the command that enables the pws returns a respose with status: CMD_STATUS_AES_DEC_FAILED

szszszsz, 2016-10-27:
Hi!
Could you post the compiler version you have tested?
I have tested this issue on current master with `gcc 5.4` and it is not occuring. I guess the cause was misused `memcpy` (https://github.com/Nitrokey/nitrokey-pro-firmware/commit/4daceed22d716a3c9433413f3b9b7ca83828490c).

MaVo159, 2016-10-27:
I suspected the same because the first wrong memcpy was exactly at this place. Didn't have time to confirm. I don't remember my gcc version. I suspect it was a bit newer than 5.4.

szszszsz, 2016-10-27:
OK, thank you. I will run tests thoroughly before closing.

szszszsz, 2016-11-12:
Issue is not occurring on latest commit. It might be also caused by not regenerating AES keys after factory reset (detailed log below).
[AES_DEC_FAILED_example.txt](https://github.com/Nitrokey/nitrokey-pro-firmware/files/587350/AES_DEC_FAILED_example.txt)

closed 2016-11-12 by szszszsz