Nitrokey/nitrokey-pro-firmware #22: Can't add AWS MFA secret

jans23, 2016-10-03:
The original ticket is here: https://github.com/Nitrokey/nitrokey-app/issues/26

szszszsz, 2016-10-27:
To sum up: issue is caused by storage limitations and needs it's structure redesign.

tripleclones, 2016-11-15:
Can you give any time frames on this please?

Just to clarify I have experienced more services using longer secret keys which are incompatible other then AWS

jans23, 2016-11-15:
No time frame yet. Which are these other services you are mentioning?

>

szszszsz, 2016-11-15:
Current TOTP secret given by AWS is 64 base32 characters which is 40 bytes (320 bits).

tripleclones, 2016-11-16:
VPS provider Bitfolk is the one that springs to mind but I'm sure there was another as well, I will check some other providers I use

jans23, 2016-11-16:
We are going to tailor our devices to AWS's secret length. It would be
important, how long the secrets of those other providers is and if such
exceed the secret's length AWS is using.

>

tripleclones, 2016-11-16:
When I set up AWS I was given a 64 Byte key. Bitfolk provided a 80 Byte key. As I understand it the Nitrokey is current limited to 32 Bytes?

szszszsz, 2016-11-16:
Hi! @tripleclones
Could you tell, are these bytes encoded base32, base64 or hex? Are there any information of their type? I guess the first, but wanted to confirm.

tripleclones, 2016-11-16:
I appears to be base32 yes, although I have asked the question to be sure.

tripleclones, 2016-11-16:
Confirmed 50 bytes base32 encoded

tripleclones, 2016-11-16:
OK so Bitfolk have drop the requirements of their key to fit in with the current 32 base32 - so its just AWS now :)

closed 2016-11-19 by szszszsz

szszszsz, 2016-11-19:
The AWS OTP keys will be supported from now on Nitrokey Pro v0.8 and hopefully on Nitrokey Storage next versions (v0.44+).

PR #29 (2016-11-19): Handle bigger OTP secrets up to 40 bytes, 320 bits
This will support AWS-length secrets - 40 bytes
Fixes #22 

Tested using libnitrokey tests under Ubuntu 16.10
[test_report_08.txt](https://github.com/Nitrokey/nitrokey-pro-firmware/files/601454/test_report_08.txt)

Summary:
- added new command for sending OTP data
- removed tables with fixed OTP slots offsets (replaced with function)
- casting slots storage and command packets to structs for easier data manipulation
- allowing for setting any maximum OTP secret size (with a cost of lower count of OTP slots over some size)
- handling 3rd HOTP s [... cut]