Nitrokey/nitrokey-pro-firmware #32: Support longer PINs

jans23, 2017-03-21:
The firmware (and NK App) support max of 20 character long PINs while the OpenPGP Card supports 32 characters. The firmware and App should be changed to support entire 32 characters.

tlaurion, 2022-06-02:
@szszszsz Was this corrected in newer firmware versions? Seems like this is related to firmware 0.10 but not 0.11?

tlaurion, 2022-06-02:
@szszszsz @jans23: the firmware (0.9, 0.10, 0.11) permits longer PINs to be provisioned.
But HOTP sealing (Heads) only works when provided with GPG Admin PIN <=25 chars.

https://github.com/osresearch/heads/pull/1168 circumvents this firmware issue under Heads OEM Factory Reset/Re-Ownership security components provisioning (forcing GPG Admin PIN <=25 chars and GPG User PIN <=64 chars) for all USB Security dongles.

szszszsz, 2022-06-10:
Hi @tlaurion! Sorry for the delay.
1. Indeed the protocol used is limited to 20-25 characters for the PIN length, depending on the command. Thank you for adding this check.
2. This problem is still present in the current Nitrokey Pro firmware unfortunately.
3. As for the solution we could: 
     - fix this problem in the current Nitrokey Pro firmware, 
     - add support to the nitrokey-hotp-verification for this change (selecting protocol by the firmware version), and 
     - leave the check from the mentioned PR only for the non-upgradable older versions, which are <= 0.10.

tlaurion, 2022-06-27:
@szszszsz So Heads now makes sure that PINs are <= 25 characters