Nitrokey/nitrokey-pro-firmware #33: Fail to generate 4096 bit keys

MauroMombelli, 2017-03-22:
Despite 4096 bit keys being said as supported, any try to generate them with GPG or GPG2 failed. The same identical setup with 2048 worked first time.

szszszsz, 2017-03-23:
Hi!
What OS and GPG version do you use? And what is the model and version of the device?

MauroMombelli, 2017-03-23:
Arch Linux with gnupg 2.1.19, notrokey with firmware 0.8.

Just to be sure, during the key creation procedure I set 4096, my data (I set no expire) and it ask me for the pins. Then it fail, it remember I tryed with 4096 but there are no keys.

One thing I notice is the company in the firmware info does not match what I saw on other example on internet

szszszsz, 2017-03-23:
> One thing I notice is the company in the firmware info does not match what I saw on other example on internet
Could you elaborate? Where did you saw it and how it looked like?

MauroMombelli, 2017-03-23:
gpg --card-status                                          

 [...]
Version ..........: 2.1
Manufacturer .....: ZeitControl
[...]

my bad seems to be correct, i got confused.

when i fail to generate the key, the 4096 size is reported in 

Key attributes ...: rsa2048 rsa2048 rsa2048

szszszsz, 2017-03-23:
> Despite 4096 bit keys being said as supported, any try to generate them with GPG or GPG2 failed. The same identical setup with 2048 worked first time.

This looks like a bug in GPG 2.1.19. We have just registered one with 2.1.15 but it would be great to have logs from the latest version.
Could you take the logs? Please follow these steps to do so: 
1. Copy [scdaemon.conf.txt](https://github.com/Nitrokey/nitrokey-pro-firmware/files/864162/scdaemon.conf.txt) to `~/.gnupg/scdaemon.conf` and correct the absolute path there to match own user home directory. If any config is already there please backup it and swap with this one.
2. Kill old `scdaemon` with `pkill scdaemon`.
3. Do not-working scenario - generate 4096 keys as usual .
4. Run step 2 again so the log file will not be appended with unneeded information by running `scdaemon` in the background.
5. Undo changes from step 1 - rename/delete or swap `scdaemon.conf` with the backup, so log files will not be generated during day-to-day use.
6. Attach the log file `scdaemon.log` here in a comment as a file.

MauroMombelli, 2017-03-24:
i would add a point
0. change your admin and user pin to a default value, as they seems to be saved in the logs :)

btw in the end it seems to fail because a libusb timeout.

[scdaemon-pro-4096.log.txt](https://github.com/Nitrokey/nitrokey-pro-firmware/files/866540/scdaemon-pro-4096.log.txt)

szszszsz, 2017-03-24:
Ah, you are right. I have forgot to mention about the PINs. Sorry!
Thank you for the logs! Indeed it looks like the timeout is a cause. I will send it further to GnuPG team.

MauroMombelli, 2017-03-29:
is there any open issue on gpg that i can follow?

jans23, 2017-03-30:
The issue is already fixed in the current development version which
should be released as GnuPG 2.1.20.

>

MauroMombelli, 2017-04-06:
Just updated GPG to .20, generation worked succesfully. Key not tested but if they will have issue I'll open a new bug report as the problem itself is solved.

Thanks!

closed 2017-04-06 by MauroMombelli

szszszsz, 2017-04-07:
Great, thank you for re-testing!