Nitrokey/nitrokey-pro-firmware #34: Key generation fails

NoiZtril, 2017-06-08:
Generation of keys fails through gpg2 --card-edit, while the key is being usable to crypt and decrypt emails. Tested on  Ubuntu 15, Fedora 24 and Debian 9.

Under Fedora24:
```
gpg/card> admin
Admin commands are allowed

gpg/card> generate
Make off-card backup of encryption key? (Y/n) Y

gpg: Note: keys are already stored on the card!

Replace existing keys? (y/N) Y
What keysize do you want for the Signature key? (4096) 
What keysize do you want for the Encryption key? (4096) 
What keysize do you want for the Authentication key? (4096) 
Please specify how long the key should be valid.
         0 = key does not expire
      <n>  = key expires in n days
      <n>w = key expires in n weeks
      <n>m = key expires in n months
      <n>y = key expires in n years
Key is valid for? (0) 3y
Key expires at Sun 07 Jun 2020 03:13:32 AM EDT
Is this correct? (y/N) y

GnuPG needs to construct a user ID to identify your key.

Real name: Testing
Email address: 
Comment: 
You selected this USER-ID:
    "Testing"

Change (N)ame, (C)omment, (E)mail or (O)kay/(Q)uit? O
gpg: key generation failed: Card error
Key generation failed: Card error

gpg/card> quit
```

gpg2 --version:
```
gpg (GnuPG) 2.1.13
libgcrypt 1.6.6
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Home: /home/user/.gnupg
Supported algorithms:
Pubkey: RSA, ELG, DSA, ECDH, ECDSA, EDDSA
Cipher: IDEA, 3DES, CAST5, BLOWFISH, AES, AES192, AES256, TWOFISH,
        CAMELLIA128, CAMELLIA192, CAMELLIA256
Hash: SHA1, RIPEMD160, SHA256, SHA384, SHA512, SHA224
Compression: Uncompressed, ZIP, ZLIB, BZIP2
```

scdaemon.log:
```
2017-06-08 02:55:02 scdaemon[7282] listening on socket '/run/user/1000/gnupg/S.scdaemon'
2017-06-08 02:55:02 scdaemon[7282] handler for fd -1 started
2017-06-08 02:55:03 scdaemon[7282] reader slot 0: using ccid driver
2017-06-08 02:55:03 scdaemon[7282] slot 0: ATR=3B DA 18 FF 81 B1 FE 75 1F 03 00 31 C5 73 C0 01 40 00 90 00 0C
2017-06-08 02:55:03 scdaemon[7282] updating reader 0 (0) status: 0x0000->0x0007 (0->1)
2017-06-08 02:55:03 scdaemon[7282] DBG: send apdu: c=00 i=A4 p1=00 p2=0C lc=2 le=-1 em=0
2017-06-08 02:55:04 scdaemon[7282] reader slot 0: using ccid driver
2017-06-08 02:55:04 scdaemon[7282] slot 0: ATR=3B DA 18 FF 81 B1 FE 75 1F 03 00 31 C5 73 C0 01 40 00 90 00 0C
2017-06-08 02:55:04 scdaemon[7282] DBG:  raw apdu: 00 A4 00 0C 02 3F 00
[... 125 lines cut ...]
2017-06-08 02:55:49 scdaemon[7282] DBG:  raw apdu: 00 CA 00 6E 00
2017-06-08 02:55:49 scdaemon[7282] DBG:  response: sw=9000  datalen=217
2017-06-08 02:55:49 scdaemon[7282] DBG:       dump:  4F 10 D2 76 00 01 24 01 02 01 00 05 00 00 3D 4A 00 00 5F 52 0A 00 31 C5 73 C0 01 40 05 90 00 73 81 B7 C0 0A 7C 00 08 00 08 00 08 00 08 00 C1 06 01 10 00 00 20 00 C2 06 01 10 00 00 20 00 C3 06 01 10 00 00 20 00 C4 07 01 20 20 20 03 03 03 C5 3C 59 24 06 2E 23 A4 BC 0B 1F 41 DA FD 94 91 70 17 80 C6 2E 11 65 C2 BD 5E 29 A0 D1 7F A1 D3 BF 94 C1 1B B3 A7 F7 D3 87 25 04 DC F1 EA 64 0A 03 4A 65 3F B7 41 0D 3E 6E 81 CA 31 36 C2 C6 3C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 CD 0C 59 37 72 3E 57 0D 31 AF 59 37 72 7D
2017-06-08 02:55:49 scdaemon[7282] existing key will be replaced
2017-06-08 02:55:49 scdaemon[7282] DBG: send apdu: c=00 i=CA p1=00 p2=C4 lc=-1 le=256 em=0
2017-06-08 02:55:49 scdaemon[7282] DBG:  raw apdu: 00 CA 00 C4 00
2017-06-08 02:55:49 scdaemon[7282] DBG:  response: sw=9000  datalen=7
2017-06-08 02:55:49 scdaemon[7282] DBG:       dump:  01 20 20 20 03 03 03
2017-06-08 02:55:49 scdaemon[7282] 3 Admin PIN attempts remaining before card is permanently locked
2017-06-08 02:55:49 scdaemon[7282] DBG: asking for PIN '|A|Please enter the Admin PIN'
2017-06-08 02:55:59 scdaemon[7282] DBG: send apdu: c=00 i=20 p1=00 p2=83 lc=20 le=-1 em=0
2017-06-08 02:55:59 scdaemon[7282] DBG:  raw apdu: 00 20 00 83 14 55 6E 64 33 72 67 72 6F 75 6E 64 20 62 69 61 74 63 68 65 73
2017-06-08 02:55:59 scdaemon[7282] DBG:  response: sw=9000  datalen=0
2017-06-08 02:55:59 scdaemon[7282] DBG:     dump:  
2017-06-08 02:55:59 scdaemon[7282] please wait while key is being generated ...
2017-06-08 02:55:59 scdaemon[7282] DBG: send apdu: c=00 i=47 p1=80 p2=00 lc=2 le=2048 em=1
2017-06-08 02:55:59 scdaemon[7282] DBG:  raw apdu: 00 47 80 00 00 00 02 B6 00 08 00
2017-06-08 02:56:12 scdaemon[7282] ccid_transceive failed: (0x1000a)
2017-06-08 02:56:12 scdaemon[7282] apdu_send_simple(0) failed: card I/O error
2017-06-08 02:56:12 scdaemon[7282] generating key failed
2017-06-08 02:56:12 scdaemon[7282] operation genkey result: Card error
2017-06-08 02:56:40 scdaemon[7282] updating reader 0 (0) status: 0x0007->0x0007 (1->2)
2017-06-08 02:56:40 scdaemon[7282] sending signal 12 to client 7280

```

szszszsz, 2017-06-08:
Hi! As far as I remember generating 4096 RSA keys works from GnuPG in version 2.1.21. It probably does not work on versions 2.1.11-20. Please try latest (or older,  like 2.0.30 or 2.1.10) GnuPG version.

szszszsz, 2017-06-08:
Closing as `invalid` since this is not an issue within `Pro` firmware and it is fixed in latest GnuPG.

closed 2017-06-08 by szszszsz