Nitrokey/nitrokey-pro-firmware #35: Support longer passphrases

jonathancross, 2017-08-15:
The current 20 character limit on passwords is too short for secure passphrases.
Would be great to support longer passwords eg: 128+ characters (especially considering Unicode chars take up more than one "character" in the password manager).

szszszsz, 2017-09-27:
Just to show current state - current memory limit for:
- OTP: 512 bytes+
- PWS: 2*512 bytes

muellermartin, 2019-01-12:
What is the current state of this issue?

I just tried out a Nitrokey Storage 2 and I'm a bit surprised that passwords can only have up to 20 characters. Unfortunately, this is not documented anywhere on the website. In my eyes, the whole purpose of such a hardware secured password manager is to store longer passphrases, which I can't.

The suggested workaround in issue Nitrokey/nitrokey-app#269 to store the password in the login field is a bit shocking to me, as this exposes the secret to bystanders and screenshots/casts. A slightly better solution would be to swap the restrictions: only 20 characters for the login and 32 characters for the password.

In my opionion, a better solution is suggested in Nitrokey/nitrokey-storage-firmware#38 to use a dynamic storage scheme, as users could choose between longer passwords/fields and fewer entries.

jans23, 2019-01-29:
We are considering to move to a master-password scheme which would have the benefits:

- No synchronization to other systems is necessary, once the master password has been setup once.
- No limit of amount of accounts (leaving meta data beside).
- Long passwords

What do you think?

jonathancross, 2019-01-30:
Can you explain more about how the _master-password scheme_ would work?

jans23, 2019-01-30:
@jonathancross See [this article](https://en.wikipedia.org/wiki/Master_Password), which covers a specific implementation too but also explains the general scheme.

jonathancross, 2019-01-31:
Thanks @jans23  Unfortunatly a master-password setup doesn't sound like it would address the original issue -- desire to use / store _passphrases_.

It still might be interesting though.

Would the system also allow users to select characters used in the generated passwords?
Some situations need special chars, others reject them, some symbols are okay, others not, sometimes first character can't be a number, etc.

jans23, 2019-01-31:
It depends on how you define "original issue". If it's defined as "securely store login credential" for instance, it would be addressed. I agree, it may require thinking outside of the box and maybe only cover 90% of use cases. But once accepted, it should serve very well.

Special charcters and options: yes.

muellermartin, 2019-01-31:
While the Master Password algorithm sounds like a nice feature, I also agree to @jonathancross that this does not solve the aforementioned problem, because people might want to simply store a pre-defined password/passphrase which currently does not fit into the current storage layout.
The Master Password feature would only allow creating new passwords, not to store exisiting ones. Therefore I'm still in favor of the dynamic storage scheme.

jonathancross, 2019-02-01:
By "original issue", I meant specifically storing passphrases (4-12 random lowercase words).  Also agree with @muellermartin about storing existing passwords (or other sensitive data like a bitcoin private key, etc.)

jonathancross, 2020-11-11:
Are _pass phrases_ (long passwords) still not supported? Or can this be closed?

szszszsz, 2020-11-12:
Hi! I am sorry, but I do not have any updates on that feature. I would like to leave it open to remind us about this particular feature demand.