Nitrokey/nitrokey-pro-firmware #38: Random data sent just after connection disrupts communication

szszszsz, 2017-10-09:
According to https://github.com/open-keychain/open-keychain/issues/1936#issuecomment-335002988 Nitrokey Pro device is sending supposedly random data sequences just after it is connected through `BulkIn` endpoint, before `IccPowerOn` is sent to initialize the connection. It could in same cases disrupt the communication process rendering the device unusable (like in Open Keychain case). 
To investigate what these bytes are and why are they sent.

> The bytes are slightly changing every time, but some are also constant:
a78ae8aaf717cbbfecc7cb88a6f6910c56418514b6aaa0b8750555142bae39823210a22bfa3f73f9278bb74b2c95f65b134452ad236ea23ad0d75617ac843d29
a78ae88af6074b3f6cc7cb88a6f7910c56418514b2aaa0b8750555042aa639823210a26bfa3f73f9a78bb74bac95f65b136450ad236ea23ad0d75617ac863d39
a78ae88af707cbbb6cc5cb8826f7d10c56418514b2aaa0ba750515142ba639a23210a22bfa3f73f1a783b74b2c91f65b134452ad336ea23ad0d75217ac843c29
Someone else reported getting these bytes:
fa2e651a767541df360a8b2a1314857c890d51c16ade92891f57051846876584caca8202f4d705878e8e81cc4d36c76d650a155d2eb8fb2d74f30af7e48f86ad

szszszsz, 2017-11-10:
Workaround for this issue has been added to OpenKeychain already, however it would be good to see what the sent data bytes are and is this conforming with the expectations (it was not expected at OpenKeychain side surely).

Valodim, 2017-11-10:
I agree. Worst case it's a dump of random memory or something like that. It was also one of the only reasons why nitrokeys didn't "just work" in OpenKeychain. Please investigate.