Nitrokey/nitrokey-pro-firmware #40: redundent warnings during compile

adlererik, 2017-11-27:
Compiling master with 

$ gcc --version
gcc (Debian 6.3.0-18) 6.3.0 20170516

Debian 9 Stable
Debian 4.9.51-1 (2017-09-28) x86_64 GNU/Linux

Will skimp though without error but there are a ridiculous amount of warnings. 95% of these warning are very easy to fix. Since I own a NitroKey Pro I actually care. If you are unwilling or do not have the time maybe I can clean the code up.

szszszsz, 2017-11-28:
Hi @adlererik !
Thank you for the report. Indeed, looking at CI there is quite a number of warnings during the compilation. Since we are busy on some other topic at the moment, could you assist? Otherwise, some time should be available at the beginning of the next year for this task.
Regarding compiler: `gcc-arm-none-eabi` is used here, which last time I have checked stopped at 4.9. Could you paste its version?

adlererik, 2017-11-29:
My arm tool chain looks like the following atm. I mostly use it for Gnuk coding but I also happen to own a NitroKey Pro and wanted to take a look at the code.

gcc-arm-none-eabi Version: 15:5.4.1+svn241155-1
gdb-arm-none-eabi Version: 7.12-6+9+b2
binutils-arm-none-eabi Version: 2.28-5+9+b3
libnewlib-arm-none-eabi Version: 2.4.0.20160527-2

Sure I can causally work on a code clean up. Like I said I own a NK Pro so I am doing everybody a favor. I’ll send it to you when I done or get bored with it.

szszszsz, 2018-04-19:
Hi @adlererik !
By any chance, have you flashed the Pro with hand-compiled firmware using gcc 5.4.1? I tried it recently for fun, but device had not started and I have not investigated this further.
Release of v0.8 is compiled with gcc 4.9.

PR #44 (2018-05-21): Remove build warnings
Remove most of the build warnings. `src/Library` files are not impacted.
Warnings generated with `-Wall`. Some of the `-Wextra` were solved too.
Now any warning in our codebase (not `src/Library`; using `-Wall switch`) would make a compilation error.
Compilation made on GCC 6.3.1 and 4.9.3. The latter build used for testing.

Fixes #40 

Tested against:
- [x] libnitrokey v3.3-12-g391a276 (`pytest -sv test_pro.py --count 3`)
- [x] nitrokey-hotp-verification v1.0 (`./test_hotp`)
commit fff97850 (2018-05-21): Merge branch 'remove_build_warnings'

Remove most of the build warnings. src/Library files are not impacted.
Warnings generated with -Wall. Some of the -Wextra were solved too.
Now any warning in our codebase (not src/Library; using -Wall switch)
would make a compilation error.
Compilation made on GCC 6.3.1 and 4.9.3. The latter build used for
testing.

Tested against:

    libnitrokey v3.3-12-g391a276 (pytest -sv test_pro.py --count 3)
    nitrokey-hotp-verification v1.0 (./test_hotp)

Closes #44
Fixes #40
closed 2018-05-21 by szszszsz