Nitrokey/nitrokey-pro-firmware #48: [Question] Has the NitroKey Pro applied for RYF Certification?

asddsaz, 2018-11-03:
**Question:**

jans23, 2018-11-03:
No, but we have OSHW certificate.

asddsaz, 2018-11-03:
@jans23 No to be rude, but you should really consider applying. The FSF has approved very few products and many people (including myself) would like to see more. You appear to meet all the requirements.

You can contact them here: licensing@fsf.org

jans23, 2018-11-03:
Smart cards in general aren't very open and wouldn't be eligible for RYF. But Nitrokey Start doesn't contain a smart card and therefore meets RYF criteria. In fact we applied Nitrokey Start for RYF but got rejected. FSF's justification was that only one certified product in the portfolio could confuse people. For me, it's a ridiculous reasoning.

closed 2018-11-03 by jans23

asddsaz, 2018-11-03:
@jans23 Thank you for the explanation.

Zig-03, 2019-02-24:
@asddsaz perhaps because of this paragraph:

> To prevent confusion among customers about exactly what product has been certified, any other products offered by the seller, which are not certified by the FSF, must be easily distinguishable from certified products: their names must not be similar and their packaging must also not be similar.

https://www.fsf.org/resources/hw/endorsement/criteria

> are not all Nitrokeys Free Software?
> Smart cards in general **aren't** very open and **wouldn't** be eligible for RYF. But Nitrokey Start **doesn't contain a smart card** and therefore meets RYF criteria.

jans23, 2019-02-25:
According to FSF, our products containing a smart card aren't eligible, even though their firmware is free software.

ncorder, 2021-11-26:
> According to FSF, our products containing a smart card aren't eligible, even though their firmware is free software.

Is there a reason the smart cards would not be RYF compliant? 

```
User installation of modified software

The seller must give the user, along with the product software source code, the practical capability to install replacement software for any and all of the free software in the device. This means the product must have the requisite facilities to install software in the processors that run free software, and include adequate and sufficient documentation on how to use them. If software is required for this installation, it is considered part of the product software.
```

@jans23 Is it not possible to update the source code on the smart cards? What segment of the [FSF's RYF criteria](https://ryf.fsf.org/about/criteria) is NitroKey not compliant with?

\-Thanks

jans23, 2021-11-27:
It's not possible to update OpenPGP Card's firmware and it's source code is not entirely published yet because APIs are under NDA.