Nitrokey/nitrokey-pro-firmware #49: Build errors / Required version of gcc-arm-none-eabi?

andreas-mausch, 2018-11-17:
Hi, what version of gcc-arm-none-eabi is required to build this?
I've tried 8.2.0 on arch and it gives me this:

```
Linking: nitrokey-pro-firmware.elf
arm-none-eabi-gcc -mthumb -mcpu=cortex-m3  -I. -gdwarf-2 -DROM_RUN -DUSE_STDPERIPH_DRIVER -DSTM32F10X_HD -DUSE_STM3210E_EVAL -DGLOBAL_VID=0x20a0 -DGLOBAL_PID=0x4108  -O2 -Wall -Wcast-align -Wimplicit -Wpointer-arith -Wswitch -Wredundant-decls -Wreturn-type -Wshadow -Wunused -Wa,-adhlns=../../src/main.lst -I../../src/inc -I../../src/stm/Libraries/CMSIS/Core/CM3 -I../../src/stm/Libraries/STM32_USB-FS-Device_Driver/inc -I../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/inc -Wno-cast-qual -Wno-cast-align -Wcast-qual -Wno-cast-qual -MD -MP -MF .dep/nitrokey-pro-firmware.elf.d    ../../src/main.o ../../src/hw_config.o ../../src/test_code.o ../../src/utils/delays.o ../../src/utils/memory_ops.o ../../src/ccid/Ccid_usb.o ../../src/ccid/Ifd_protocol.o ../../src/ccid/Crd.o ../../src/ccid/Ifd_ccid.o ../../src/ccid/CcidLocalAccess.o ../../src/ccid/smartcard/smartcard.o ../../src/ccid/CCIDHID_USB/CCIDHID_usb_prop.o ../../src/ccid/CCIDHID_USB/CCIDHID_usb_desc.o ../../src/sd-disk/sd-usb/usb_desc.o ../../src/sd-disk/sd-usb/usb_prop.o ../../src/stm/stm32f10x_systick.o ../../src/stm/stm32f10x_it.o ../../src/stm/Libraries/CMSIS/Core/CM3/core_cm3.o ../../src/stm/Libraries/CMSIS/Core/CM3/system_stm32f10x.o ../../src/stm/Libraries/CMSIS/Core/CM3/startup/gcc/startup_stm32f10x_hd.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/misc.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_adc.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_bkp.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_can.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_crc.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_dac.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_dbgmcu.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_dma.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_exti.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_flash.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_fsmc.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_gpio.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_i2c.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_iwdg.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_pwr.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_rcc.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_rtc.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_sdio.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_spi.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_tim.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_usart.o ../../src/stm/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_wwdg.o ../../src/stm/Libraries/STM32_USB-FS-Device_Driver/src/usb_core.o ../../src/stm/Libraries/STM32_USB-FS-Device_Driver/src/usb_init.o ../../src/stm/Libraries/STM32_USB-FS-Device_Driver/src/usb_int.o ../../src/stm/Libraries/STM32_USB-FS-Device_Driver/src/usb_mem.o ../../src/stm/Libraries/STM32_USB-FS-Device_Driver/src/usb_regs.o ../../src/crypt/aes/aes.o ../../src/crypt/sha1/sha1.o ../../src/crypt/sha1/hmac-sha1.o ../../src/hotp/hotp.o ../../src/keyboard/keyboard.o ../../src/keyboard/report_protocol.o ../../src/usb/usb_endp.o ../../src/usb/usb_bot.o ../../src/usb/usb_pwr.o ../../src/usb/usb_istr.o ../../src/pwd-safe/FlashStorage.o ../../src/pwd-safe/HandleAesStorageKey.o ../../src/pwd-safe/password_safe.o     --output nitrokey-pro-firmware.elf -nostartfiles -Wl,-Map=nitrokey-pro-firmware.map,--cref -lc  -lm -lc -lgcc  -lrdimon    -Tstm32.ld
/usr/lib/gcc/arm-none-eabi/8.2.0/../../../../arm-none-eabi/bin/ld: cannot find -lc
/usr/lib/gcc/arm-none-eabi/8.2.0/../../../../arm-none-eabi/bin/ld: cannot find -lm
/usr/lib/gcc/arm-none-eabi/8.2.0/../../../../arm-none-eabi/bin/ld: cannot find -lc
/usr/lib/gcc/arm-none-eabi/8.2.0/../../../../arm-none-eabi/bin/ld: cannot find -lrdimon
/usr/lib/gcc/arm-none-eabi/8.2.0/../../../../arm-none-eabi/bin/ld: cannot find -lg
/usr/lib/gcc/arm-none-eabi/8.2.0/../../../../arm-none-eabi/bin/ld: cannot find -lc
collect2: error: ld returned 1 exit status
```

I'm considering buying a Nitrokey but it's a requirement for me that I can build the firmware myself.

andreas-mausch, 2018-11-17:
nvm, found it in the travis.yml: Version is 4.9

szszszsz, 2019-01-04:
Hi @andreas-mausch !
Correct, it is 4.9. This is valid issue nevertheless - the used compiler version should be mentioned in the Readme. Reopening to track the change.