Nitrokey/nitrokey-pro-firmware #53: Firmware for a general purpose smart card reader

persmule, 2019-01-07:
Recently my nitrokey pro's shell has cracked, with the hardware inside undamaged. I noticed with the openpgp card removed, the nitrokey pro even refused to be enumerated as a usb device, and on hsm branch there is a commit to `Set USB serial to HSM smartcard's one` ( 3bf8a9445485785ae821c5293d74aa637ce659b7 ), which may mean a nitrokey pro with an hsm firmware cannot work without a smartcard hsm inserted in. Is there any branch to power the hardware as a general purpose smart card reader (e.g. enable it to read other smart cards, such as GSM's SIM) ?

szszszsz, 2019-01-07:
Hi!
Indeed, Nitrokey Pro is not booting up without the smart card, and the same is with HSM as far as I am concerned. During the boot process, the device negotiates the communication channel with the smart card, and perhaps then stops execution, if it was not possible.

It is hard to say for me, will GSM's SIM would work, since it may not response in the expected way, as with OpenPGP or HSM. The new card has to support `T=1` protocol as well.
I would definitely check the HSM branch (which aims to be a reader-only firmware), and see what happens. It would need to be modified though, by removing all local access (like requesting a serial number) to the smart card.

szszszsz, 2019-08-12:
Closing as replied. Please reopen / comment if otherwise.

closed 2019-08-12 by szszszsz