Nitrokey/nitrokey-pro-firmware #66: Support automatic updates via fwupd

4jNsY6fCVqZv, 2019-10-05:
Could you make that happen?
for reference please take a look at https://github.com/Nitrokey/nitrokey-storage-firmware/issues/78

jans23, 2019-10-07:
Currently a firmware update of Nitrokey Pro requires physical access to the PCB (electronics). We plan to release a revision next year which will allow signed firmware updates and potentially support fwupd too.

4jNsY6fCVqZv, 2019-10-08:
Thanks for your answer. However, revision means that users must purchase a new device and existing devices cannot enjoy the benefits, right?

jans23, 2019-10-08:
That's right.
-- 
Sent from a mobile device.

4jNsY6fCVqZv, 2019-10-09:
Too bad, and thank you for your answer!

4jNsY6fCVqZv, 2019-10-09:
Too bad, and thank you for your answer and that you are working on an integration for future versions!

ncorder, 2021-11-26:
> That's right. -- Sent from a mobile device.

Hey Jans23, is it then possible for users to verify the firmware running on the device currently?

jans23, 2021-11-27:
Nitrokey Pro's firmware can be updated via software but not verified. I think the firmware builds mostly reproducible.