Nitrokey/nitrokey-pro-firmware #68: PGP and S/MIME simultanous use

szszszsz, 2019-12-28:
Feature request to use both PGP and S/MIME together, simultaneously. 
Ticket for idea tracking. Since this is not a task for firmware implementation, tagging as `invalid`.

Details: https://support.nitrokey.com/t/pgp-smime-simultaneously-with-nk-pro-2/2100/3

alex-nitrokey, 2020-01-02:
This is already working with OpenSC and GnuPG. The only downside is that you can not (at least easily) use slot 3 (authenticate) for GnuPG stuff. But OpenSC can nearly always used as replacement.

Scenario:
Slot 1 = GnuPG key with SC usage flag
Slot 2 = GnuPG key with E usage flag
Slot 3 = S/MIME key + Cardholder certificate

The linked issues are partly misunderstood imho because they are about extending OpenSC based on new hardware possibilities but does not help for the PGP/S/MIME use case.

alex-nitrokey, 2020-04-06:
actually, I think this ticket can be closed

jans23, 2020-04-06:
Is there any documentation describing how to initialize a Nitrokey to use it for both GPG and S/MIME? I believe this should be created before closing this ticket.