Nitrokey/nitrokey-pro-firmware #73: Nitrokey Pro firmware linker files ambiguously licensed

jeremiah, 2020-06-08:
Hi,

At Purism we use the Nitrokey Pro as the basis for our own key called Librem Key. We're submitting the Librem Key for [Respects Your Freedom certification](https://ryf.fsf.org/). In discussions with the certification folks they note that there's some ambiguity regarding the headers in the files in the nitrokey-pro-firmware/build/gcc directory.

The files STM32_COMMON.ld & STM32_SEC_FLASH.ld have a header which says;
```
/*
Common part of the linker scripts for STR71x devices in FLASH mode
(that is, the FLASH is seen at 0)
Copyright RAISONANCE 2005
You can use, modify and distribute thisfile freely, but without any waranty.
*/
```
while STM32_128K_20K_FLASH.ld and stm32.ld have header that say;
```
/*
Linker script for STM32F10x
Copyright RAISONANCE 2007 (modified by Lanchon 1-Feb-2008)
You can use, copy and distribute this file freely, but without any waranty.
Configure memory sizes, end of stack and boot mode for your project here.
*/
```
The issues are;
-- The word "warranty" is spelled "waranty". 
-- Not all the files explicitly grant the right to modify.
-- Not all the files explicitly grant the right to copy.

I'd like to propose a PR which I hope might fix this and be suitable for your needs. Alternatively, those files can be fixed by whomever wrote them if they still hold the copyright and have not contributed them into the public domain.

szszszsz, 2020-06-09:
Hi!
Sure! Please take a look into it. Alternatively linker files could be rewritten, as I do not expect any special magic there.

jeremiah, 2020-06-09:
Good point, I'll take a look. So far we're adding our VID so we are modifying the file but that is likely not of interest for you folks (as you have different data). I don't think we're doing anything fancy, just reusing the files but if we modify them enough perhaps we can just put them under the GPL for our use (which would make the FSF happier. :-)

jeremiah, 2020-10-08:
Gentle bump for review. :-)

szszszsz, 2020-10-12:
Hi!  Will look at this tomorrow. Sorry for the delay!