Nitrokey/nitrokey-pro-firmware #75: Support more OTP slots

szszszsz, 2020-06-30:
Support more OTP slots, e.g. 32 or more.

Details: https://support.nitrokey.com/t/how-many-totp-hotp-secrets-can-a-nitrokey-store/2445/3

ELLIOTTCABLE, 2021-09-27:
As I'm coming up on 32 creds on my *second* yubikey (so, 64 in total), I'm running into this frustrating fucking wall again.

Just bumping this, in hopes that someone might be convinced "hey, if we support 128 creds or something, we might win some customers over from Yubico!"

Please. It's 2021. WebAuthn is still *years* if not a whole *decade* off from genuinely replacing TOTP; and more companies are adding TOTP (not WebAuthn! Still TOTP!) requirements every single day. /=

jans23, 2023-08-21:
We won't increase this limit for Nitrokey 3 but we recommend Nitrokey 3 instead which can be used with 50 OTP slots. Technically this amount can be increased. Please submit another issue ticket in [Nitrokey 3 repository](https://github.com/Nitrokey/nitrokey-3-firmware/) if you are interested to increase this limit.

closed 2023-08-21 by jans23