Nitrokey/nitrokey-pro-firmware #84: Customise firmware

sieveo, 2021-06-11:
Hi there!

We are interesting in using your Nitrokey HSM for some software projects, but we have the requirements of enabling the HSM with crypto that is yet to be standardised  - in particular Pairing based ECC BN from https://github.com/miracl/core

We are happy with doing the work ourselves, and from what I can see here it sounds like we could modify the firmware to add this type of functionality, but I am not 100% sure about it. Is this possible in theory to add this type of functionality and have private keys protected in a device like the Nitrokey HSM and have the HSM do signing?

jans23, 2021-06-11:
In general, the entity which protects cryptographic keys also needs to compute the algorithm. Because smart cards don't support your algorithms yet, the only option is to add support to the firmware which runs on the MCU. I strongly recommend to do this on the [Rust-based platform Trussed ](https://github.com/trussed-dev/trussed) of our upcoming Nitrokey 3. Your contributions are very much welcome.

Because this ticket belongs to a different project, I'm closing it.

closed 2021-06-11 by jans23

jans23, 2021-06-11:
Here is a [trial wrapping](https://github.com/trussed-dev/trussed/pull/12) already.

sieveo, 2021-06-11:
@jans23 that is excellent thank you! I'll have a look at it. I'm not really familiar with Rust, but I've only heard good things so far and coming from C/C++ it seems like it is not too much of a leap.

Trussed looks really interesting. :)