Nitrokey/nitrokey-pro-firmware #85: Adding Curve25519 support (on my own)

nothien, 2021-06-18:
Hi!  All my personal PGP keys use Curve25519.  It seems that for the Nitrokey Pro (and Storage), only NIST P and brainpool are going to be supported, but I am still interested, since the firmware is open-source.  I'm wondering if it's feasible for me to implement Curve25519 support in the firmware and update a Nitrokey with it, given that:
* The Nitrokey Start firmware provides Curve25519 support, and it should be possible to reuse that code.
* I have previous experience with C and systems programming.
In particular, are there any tips or pointers you can provide for starting out with the codebase?  Things like what modules to look into, notes on uploading custom-written firmware, etc.
Thank you!

szszszsz, 2021-06-21:
Hi @nothien!
This might be a bit harder, than it looks. In Nitrokey Pro and Nitrokey Storage we use an OpenPGP v3 smart card for storing and handling the key material, and not controlling the CCID traffic at all, except for being a relay for the traffic.  The smart card does not handle the Curve25519 because of the missing primitives implementation for this algorithm, and we have learned that this is so far not in the vendor's roadmap.
In theory it should be possible to add software support for the Curve25519, with hijacking the CCID traffic by replying to Curve25519 related queries from the firmware instead of the smart card, but this might be too much work for the effect. 
It might be easier to begin with Nitrokey Start firmware (since it has the queries interpreter already, as it is a complete software implementation), and send the RSA and other features related queries to the smart card, while processing the Curve25519 locally (and have protected storage, eg. encrypted through AES key required by PIN, as it is done with Password Safe).

I plan to document the Nitrokey Pro implementation in the coming days - I will write here with an update. Just following the `main()` function should answer most of the questions.
Regarding Nitrokey Pro / HSM / Start flashing you should find all the information here:
- https://github.com/Nitrokey/nitrokey-pro-firmware#flashing

Let me know if you would like to work with Nitrokey Storage (Nitrokey Pro platform should be a better choice to start though).

To sum up, this would be a very nice addition to Nitrokey Pro and Nitrokey Storage models, however might not be feasible while starting from their current state. Instead, I would encourage you to join development for the upcoming Nitrokey 3 written in Rust:
- https://github.com/Nitrokey/solo2 (temporary repository)

From the overview we plan to do there something similar, that is offload most of the operations to the secure element, while handling the rest in the software. The Curve25519 support is on the roadmap AFAIR.

nothien, 2021-06-22:
Thank you for the detailed response!
Although I'd heard of the Nitrokey 3, I hadn't realized that Curve25519 is a planned feature (it's mentioned all over the device description).  I'll look into the firmware repo, and contribute if I can.

closed 2021-06-22 by nothien