Nitrokey/nitrokey-pro-firmware #87: Instructions with modern OpenOCD?

saper, 2021-08-23:
One of the OpenOCD forks listed is dead and in the another one can't find the `vsllink-swd.cfg` file. But it seems that OpenOCD itself supports SWD for quite a long time (just tried 0.11.0 release with some other ARM Cortex-M chip and it worked).

Maybe just using `transport select swd` would work?

(don't have a SWD-enabled token yet to try)

szszszsz, 2021-08-24:
Hi!
You are right - the instructions here are a bit outdated at the moment. I will update them a bit. In the meantime, the recent OpenOCD is working just fine, and in fact it is used in one of the development helpers [1] available in the repo.
Hopefully the development guide at [2] will be more resourceful too.

[1] https://github.com/Nitrokey/nitrokey-pro-firmware/blob/master/Makefile#L52
[2] https://github.com/Nitrokey/nitrokey-pro-firmware/blob/master/DEVELOPMENT.md

szszszsz, 2021-08-24:
Corrected in 9e2489bbd03275eb9be74e88b70aa923c8ffc335. Closing as done.

closed 2021-08-24 by szszszsz

saper, 2021-08-24:
Wow! Perfect, this is was fast. I think we can remove https://github.com/Nitrokey/nitrokey-pro-firmware/blob/master/scripts/flash-versaloon.sh from both branches I guess...

szszszsz, 2021-08-25:
Indeed, we had not used it for a while. Actually, this script should be valid when used `stlink-v2` instead of the versaloon provided right now. Will keep in mind to correct that next time. PR's are always welcomed :-)