Nitrokey/nitrokey-pro-firmware #91: OTP functionality

jerabaul29, 2022-01-23:
Would there be a possibility, either in the nitrokey pro or in a future / separate product, to include OTP generation?

szszszsz, 2022-01-24:
Hi!
TOTP/HOTP codes generation is already integrated with Nitrokey Pro and Nitrokey Storage. Nitrokey App is required to handle this feature (or any libnitrokey client). By numbers there are 15 TOTP slots (14 on Nitrokey Storage currently), and 3 HOTP slots.
We have it on roadmap for Nitrokey 3 as well.

For details see these pages in documentation:
- https://docs.nitrokey.com/pro/windows/2fa-google.html
- https://docs.nitrokey.com/pro/windows/2fa-microsoft.html
- https://docs.nitrokey.com/pro/windows/2fa-nextcloud.html

Please reopen in case this would not answer your question.

closed 2022-01-24 by szszszsz

jerabaul29, 2022-01-24:
Mmmh, is it really implemented on the hardware token if there is need for an app? Ie is the app only for communicating with the key, or does the app do part or all of the work?

szszszsz, 2022-01-24:
App is needed for communication only. Shared secret storage and OTP code calculation are done on the device.

jerabaul29, 2022-01-24:
Aaah, thanks, I did not know about this :) .

My 2 cents: a bit sad that a custom app is needed to communicate; having a simple universal command line tool that could hook to any hardware token of this style with a general API would be even more exciting. In the same way that no app is needed to do RSA encryption / decryption, as the token implements the GPG smartcard API :) . Do you know if such a standard exists?

szszszsz, 2022-01-24:
Sadly I believe there is no open specification for implementing API for OTP, hence everyone is doing their proprietary implementations for communication. 
We do have a command line application if you are interested (Rust; multi-platform):
- https://github.com/d-e-s-o/nitrocli/tree/master/#nitrocli

szszszsz, 2022-01-24:
We are working on Nitrokey Webcrypt at the moment, which hopefully could introduce such standard, among others:
- https://github.com/Nitrokey/nitrokey-webcrypt#nitrokey-webcrypt