Nitrokey/nitrokey-pro-firmware #99: PGP key used to sign binary firmware is not available

jjakob, 2023-05-22:
The released precompiled binary firmware is signed with a PGP key but there is no information on where to obtain the key. I tried a bunch of public keyservers but none had it. It's not possible to verify the signature without the key.
The key should be published on public keyservers. Ideally it should also be signed by other people to make a web of trust possible.

szszszsz, 2023-05-23:
Hi! Thank you for the report.
Corrected: added second signature (using `868184069239FF65DE0BCD7DD9BAE35991DE5B22`), and uploaded original signature's key (`1A1A32875348A792ADA7BD6D878F36EEEA6A8D00`) to the Ubuntu's keyserver - should be synchronized with others as well.

jjakob, 2023-05-23:
Thanks, I was able to receive the key from hkps://keyserver.ubuntu.com and verify the signature.