Nitrokey/nitrokey-pro-firmware #100: Nitrokey HSM: handle v4.0 smart cards

szszszsz, 2023-11-26:
The new H/W v4.0 HSM smart cards are not properly setting up with the current firmware. The problem with communication starts right away after execution of the PPS procedure, where the higher bit rate is configured with the smart card, after which the USART peripheral working in the SMARTCARD mode is reconfigured with higher bitrate as well. After this point smart card is requested for its serial number, which ends up malformed. Similarly further USB CCID requests, which are passed to the smart card, are either ending up in a Default Handler due to the Frame Exception being raised, or just in an infinite loop, where the MCU awaits for the flag change from the USART peripheral, which at some point should signalize data reception.
Interestingly, when the `PTS_config()` body execution is skipped (= insert `return` as a first line), everything works fine at the default speed of `9600bps`, however this speed is clearly not enough to have a proper user experience. The aim is to have at least the speed of the previous smart card generation, which is `~ 115kbps`.
Even if the configuration of the minimal speed is attempted, which should end up in the same value as the default one - `9600bps`, the proper communication is not achieved.

The v3.0 cards work properly with the current implementation. The only changes are:
- ATR change, allowing for a higher bitrate;
    - old ATR: https://smartcard-atr.apdu.fr/parse?ATR=3b+de+18+ff+81+91+fe+1f+c3+80+31+81+54+48+53+4d+31+73+80+21+40+81+07+1c
    - new ATR: https://smartcard-atr.apdu.fr/parse?ATR=3B+DE+96+FF+81+91+FE+1F+C3+80+31+81+54+48+53+4D++31+73+80+21+40+81+07+92
- hardware change - JCOP 3 -> JCOP 4.

Done so far:
- [x] rewritten PTS procedure, and verified execution
- [x] tried different T0 values
- [x] tried previous generation ATR directly
- [x] verified USB CCID descriptors
- [x] compared the current smartcard HAL with the latest available STM32 examples

Things to check yet:
- [ ] peripheral microcode update
- [ ] clock dependencies (USB, smart card)
- [ ] higher peripheral clock setup (different prescaler) for the smaller bitrate error (for the `225kbps` setting)
- [ ] ETU configuration, and connected hardcoded delays adjustments (new ATR has half of the previous ETU)

Initial configuration:
- function to be filled in a minute

Mentioned `PTS_config()`:
https://github.com/Nitrokey/nitrokey-pro-firmware/blob/47dd7f1ad6789e5b317266b831d8624c7d126063/src/ccid/smartcard/smartcard.c#L507-L670

eberkhan, 2025-05-26:
I recently received two Nitrokey HSM2s. One of them is more or less working stably (not always!), the second one tends to stop responding after only a few requests, maybe related to temparature a little bit. I found this issue and decided to have a look myself.

So I set up a NUCLEO-F103RB with the card from the less stable Nitrokey HSM2. Without any further changes (just adopting some clock values due to 8MHz main clock vs. 12MHz on the Nitrokey ...), I found the same behavior: card only responded up to the modified 84375 baud rate.

Then I had a look on the I/O line with a scope. What I have found so far may be a missing pull up resistor on the smart card I/O line. The rising edges of the serial signal sent by the STM32 are just too flat for the high bit rates without this pull up. This cannot be seen for the smart card responses (card may be push-pull driving the line while sending actively). I cannot see such a pull-up in the v2 schematics that are available, nor can I measure one on the v4 hardware I have. So, unless you added a switchable pull up in the v4 version, there does not seem to be one.

Adding a 1k pull up on the I/O line changes the game. The card works flawlessly at 4.5 MHz card clock and 281250 baud. Signals look much better.

Unfortunately, this does seem to be very helpful for existing hardware ...

What do you think, can this be the reason for the flaw? (Maybe older cards have an internal pull up?)

eberkhan, 2025-05-27:
[image]

After being confident enough regarding the improvement, I applied a similar fix to the Nitrokey itself (4.7k 0603 from card holder Vcc to I/O). Seems to be more stable now, e.g., calling `sc-hsm-tool` after an interrupted `find` command in `opensc-explorer` works after some time (card reset?). This was never possible before. Firmware of course not changed.